rbei-etas/busmaster #950: The file "mfc120.dll" is missing as startup

PPSXIN, 2016-07-05:
Hello,
I have installed v3.0.0, but unfortunately, while I tried to start up the software, the warning is propelled, I have attached the warning capture FYR. BTW, my PC OS is win7 X64.
[image]

RBEI-Prathiba, 2016-07-07:
BUSMASTER v3.0.0 requires Visual Studio 2013 Redistributable installation.
For installation, please refer the link https://github.com/rbei-etas/busmaster/wiki/Visual-Studio-2013-Redistributable

Gardneia113, 2016-07-14:
[image]

I have installed Visual Studio 2013 Redistributable,but the interval of sending message, is the twice of my setting.

RBEI-busmaster, 2016-11-30:
Issue fixed with v3.1.0

closed 2016-11-30 by RBEI-busmaster