rbei-etas/busmaster #951: The new transmit interface doesn't interpret enumerate

roblamoreaux-swri, 2016-07-06:
The old transmit interface would turn the numbers into the signal description strings with a drop down box to select from my signals with enumerated values. The new one does not. I really miss this since it is very hard to remember what number is what command. Is this coming back?

In the mean time I am going back to 2.6.3

astrand, 2016-07-07:
Do you mean the J1939 TX Dialog?

roblamoreaux-swri, 2016-07-07:
This is the 2.6.3 transmit dialog. The Physical value shows my enumerated values strings:

[cid:image003.jpg@01D1D844.DC217150]

This is the 3.0 Transmit Dialog from the same configuration. Note that the physical value is the decimal value of the signal not the string.

[cid:image005.jpg@01D1D844.DC217150]

Rob LaMoreaux
Principal Engineer
Ann Arbor Technical Center
Southwest Research Institute
Engine, Emmissions and Vehicle Research Division
2001 Commonwealth Blvd. Suite 202, Ann Arbor, MI 48105
Cell: 734-972-6023 Fax: 734-913-0696
email: robert.lamoreaux@swri.org
Visit us at: www.swri.edu
"Benefiting government, industry and the public through innovative science & technology."
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 The information in this email may be proprietary and may contain protected business information. This email is intended to be reviewed and used only by the individual or organization named above. If you are not the intended recipient or an authorized representative of the intended recipient, you are hereby notified that any review, dissemination or copying of this email and its attachments, if any, and any use or disclosure of the information contained herein, is prohibited. If you have received this email in error, please immediately notify the sender by email or telephone and permanently delete this email from your system.
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

jasharv, 2017-04-05:
Is there an ETA on this issue? From what I can see in the source code the functionality is still there but i guess there is some issue with the new database handling?

ameliip, 2017-10-17:
Same problem here. State encoded data does not get interpreted also in main CAN windows. Any idea on *when* this will be fixed?

Thanks!

loki980, 2018-06-15:
We're evaluating this tool as an alternative to others we use.  This is one of the only shortcomings I've found.  It'd be nice to have this resolved or at least have DBManager returned to open source so the community can do so!

trripy, 2021-05-14:
Any update on this team ? It would be really nice to have the enum working .

GiuseppeReefilla, 2024-03-11:
Here in 2024 and this issue is still there!

cburlacu, 2025-01-03:
https://github.com/rbei-etas/busmaster/commit/637306a7ac9e05c5b3549ae6ee3855901052cab1