rbei-etas/busmaster #952: Busmaster Log to ASC conversion problem

tuvok78, 2016-07-08:
Hi, I have a problem with this kind a conversion.
In the result asc file there is just the header, no datas.
It seems as there is nothing logged inside the source file.
But the source file is full of data.

this is the result asc file:

date Fri Jul 8 4:20:51 pm 2016
base hex  timestamps absolute
no internal events logged
// version 7.1.0
End TriggerBlock

RBEI-Srinibas, 2016-07-12:
Please share the log file to BUSMASTER@in.bosch.com

RBEI-busmaster, 2016-07-29:
Format Converter utility allows .log to .asc conversion for CAN. But the .log file shared is of J1939 protocol. Currently, J1939 .log to .asc conversion and vice versa is not supported.

RBEI-Vijay, 2017-12-13:
Is it possible to BULK convert LOG to ASC files? If not through GUU, at least via command line

Sagnik-EV, 2021-08-12:
Anyone having any idea on this ? I have a good number of .log files which are need to be converted into .asc format. Currently using format converter I can manually select one file at a time for conversion. 
Can it be using some python/vbs script which basically calls this function in a loop ?