rbei-etas/busmaster #957: problem with compilling BusMaster 3.0.0 sources

mathees1, 2016-07-13:
I downloaded source code of BusMaster v3.0.0 and opened it in MS Visual Studio 2013 Community. I tried to compile it with order showed in Development Environment document:

1. Kernel\BusmasterKernel.sln
2. DBManager\DBManager.sln
3. BUSMASTER\BUSMASTER.sln
...

I cannot find DBManager folder and while trying build solution BUSMASTER.sln there are some files missing.

Should new version of BusMaster be compiled in some other way? Or do you have sources of previous version available?

Best regards
Mateusz

rileyshaw, 2016-07-22:
Same problem, disappointed to see no response in 9 days. After googling DBManager\DBManager.sln I found that Busmaster used to have a folder within the repo called DBManager but the link is now broken.

rileyshaw, 2016-07-22:
I was able to get a successful build when changing to Release instead of Debug and simply skipping the DBManager step. Hope that helps you at least get a build running.

RBEI-busmaster, 2016-07-25:
Updated build.bat is available with the latest commit 'b72911490baa42aa611bfff5f8a4cfa4808e6b7a'. Please use the latest build.bat available at 'https://github.com/rbei-etas/busmaster/tree/master/Sources' for a successful build.