rbei-etas/busmaster #958: New driver implementation

BuddyX10, 2016-07-13:
Hi,

I was trying to put my own driver on Driver Selection under CAN selection but I encountered problems. According to the instruction I used DYNAMIC LINK LIBRARY : BusmasterDriverInterface where I implemented my driver and whole project compile without errors.
Unfortunately when I used build.bat to build whole program I recived a lots of errors.

Should I do something else than I already done? Should I implement driver other method?

Thanks for answer,

RBEI-busmaster, 2016-07-25:
Please use the latest build.bat available at 'https://github.com/rbei-etas/busmaster/tree/master/Sources' for a successful build.

nabiali1994, 2017-08-16:
Hello, It's necessary to build with VS2013 ?