rbei-etas/busmaster #967: Busmaster .log to .asc conversion bug (timestamps)

CANberryy, 2016-07-29:
Hi,
I noticed another bug in the conversion of .log into .asc files with the implemented Konverter. 
The timestamps in the .log file don't match the timestamps in the .asc file, when exceeding 1 hour.
After every hour (3600s, 7200s, ...) in the .log file the konverted result is +60s ( 3660s, 7260s +60s = 7320s, ...)

I checked it with several .log-files, it's always the same. Doesn't matter if one large .log file or several parts.

Example:
.log-file:
...
00:59:59:9905 Rx 1 0x306 s 8 8E 2A C4 2A BD 2A A7 2A 
00:59:59:9907 Rx 1 0x307 s 8 CD 2A D7 2A BC 09 C6 09 
00:59:59:9919 Rx 1 0x390 s 8 BC 09 C6 09 00 00 BC E5 
01:00:00:0000 Rx 1 0x306 s 8 8F 2A C4 2A BD 2A A8 2A 
01:00:00:0002 Rx 1 0x307 s 8 CE 2A D7 2A C4 09 CD 09 
01:00:00:0014 Rx 1 0x390 s 8 C4 09 CD 09 00 00 C6 E5 
***END DATE AND TIME 26:7:2016 18:25:22:312***
***[STOP LOGGING SESSION]***

=> .asc-file:
date Tue Jul 26 5:25:22 pm 2016
base hex  timestamps absolute
no internal events logged
// version 7.1.0
...
3599.990500 1 306 Rx d 8 8E 2A C4 2A BD 2A A7 2A 
3599.990700 1 307 Rx d 8 CD 2A D7 2A BC 09 C6 09 
3599.991900 1 390 Rx d 8 BC 09 C6 09 00 00 BC E5 
3660.000000 1 306 Rx d 8 8F 2A C4 2A BD 2A A8 2A 
3660.000200 1 307 Rx d 8 CE 2A D7 2A C4 09 CD 09 
3660.001400 1 390 Rx d 8 C4 09 CD 09 00 00 C6 E5 
End TriggerBlock

WillieShih, 2016-08-24:
I have same problem.
I found the source code Log_Asc_Parser.y have error.

https://github.com/rbei-etas/busmaster/blob/master/Sources/BUSMASTER/Format%20Converter/LogAscConverter/Log_Asc_Parser.y

The fourteenth row of the function nGetAscTimeStamp
original
unTotalSec = nHours*3660 + nMin * 60 + nSec;
correct
unTotalSec = nHours*3600 + nMin * 60 + nSec;

Please release the Patch update this issue.
Thanks

CANberryy, 2016-08-25:
Yea that's it! Thank you very much!

RBEI-busmaster, 2016-11-30:
Fix is available as part of v3.1.0

closed 2016-11-30 by RBEI-busmaster