rbei-etas/busmaster #972: _Unwind_Resume undefined reference

alberto77, 2016-08-02:
Dear all,
I am trying to compile .cpp file for node simulation. I run Busmaster v3.0.0 and TDM gcc installed according to Busmaster documentation. I got the following error message:

- test.o:test.cpp <omissis> undefined reference to _Unwind_Resume.

Serching the Internet I found some issue about the use of latter gcc version, greater than 4.8.1. Unfortunately default TDM installation is bounded to gcc version 5.1.0. I unstalled TDM and modified the default installation set-up to force gcc 4.8.1 installation.

It was still not possible to compile, but I got different errors:
- undefined reference to __gxx_personality_v0;
- undefined reference to _Unwind_sjLj_Register;
- undefined reference to _Unwind_sjLj_Unregister;

Hence I tried with MinGw installation. Right now, latter version of MinGw is greater is 4.9.3 (I do not remember, I unstalled it because I had a lot of problem even in compilig trivial cpp files.

Eventually, I decided to restore original configuration installing TDM gcc with default installation setup and gcc path stored in PATH environment variable.

For sake of clarity, this is a compilation of a trivial .cpp file with -v option:
`Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=C:/TDM-GCC-32/bin/../libexec/gcc/mingw32/5.1.0/lto-wrapper.exe
Target: mingw32
Configured with: ../../../src/gcc-5.1.0/configure --build=mingw32 --enable-languages=ada,c,c++,fortran,lto,objc,obj-c++ --enable-libgomp --enable-lto --enable-graphite --enable-libstdcxx-debug --enable-threads=posix --enable-version-specific-runtime-libs --enable-fully-dynamic-string --enable-libstdcxx-threads --enable-libstdcxx-time --with-gnu-ld --disable-werror --disable-nls --disable-win32-registry --disable-symvers --enable-cxx-flags='-fno-function-sections -fno-data-sections -DWINPTHREAD_STATIC' --prefix=/mingw32tdm --with-local-prefix=/mingw32tdm --with-pkgversion=tdm-1 --enable-sjlj-exceptions --with-bugurl=http://tdm-gcc.tdragon.net/bugs
Thread model: posix
gcc version 5.1.0 (tdm-1) 
COLLECT_GCC_OPTIONS='-v' '-mtune=generic' '-march=pentiumpro'
 C:/TDM-GCC-32/bin/../libexec/gcc/mingw32/5.1.0/cc1plus.exe -quiet -v -iprefix C:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/ -D_REENTRANT hello.cpp -quiet -dumpbase hello.cpp -mtune=generic -march=pentiumpro -auxbase hello -version -o C:\Users\Admin\AppData\Local\Temp\ccHloCCy.s
GNU C++ (tdm-1) version 5.1.0 (mingw32)
	compiled by GNU C version 5.1.0, GMP version 4.3.2, MPFR version 2.4.2, MPC version 0.8.2
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "C:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/../../../../mingw32/include"
ignoring duplicate directory "C:/TDM-GCC-32/lib/gcc/../../lib/gcc/mingw32/5.1.0/include/c++"
ignoring duplicate directory "C:/TDM-GCC-32/lib/gcc/../../lib/gcc/mingw32/5.1.0/include/c++/mingw32"
ignoring duplicate directory "C:/TDM-GCC-32/lib/gcc/../../lib/gcc/mingw32/5.1.0/include/c++/backward"
ignoring duplicate directory "C:/TDM-GCC-32/lib/gcc/../../lib/gcc/mingw32/5.1.0/include"
ignoring duplicate directory "C:/TDM-GCC-32/lib/gcc/../../lib/gcc/mingw32/5.1.0/../../../../include"
ignoring duplicate directory "C:/TDM-GCC-32/lib/gcc/../../lib/gcc/mingw32/5.1.0/include-fixed"
ignoring nonexistent directory "C:/TDM-GCC-32/lib/gcc/../../lib/gcc/mingw32/5.1.0/../../../../mingw32/include"
#include "..." search starts here:
#include <...> search starts here:
 C:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/include/c++
 C:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/include/c++/mingw32
 C:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/include/c++/backward
 C:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/include
 C:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/../../../../include
 C:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/include-fixed
End of search list.
GNU C++ (tdm-1) version 5.1.0 (mingw32)
	compiled by GNU C version 5.1.0, GMP version 4.3.2, MPFR version 2.4.2, MPC version 0.8.2
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: 2e47d181020f054da881eda9fc3e9f75
COLLECT_GCC_OPTIONS='-v' '-mtune=generic' '-march=pentiumpro'
 C:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/../../../../mingw32/bin/as.exe -v -o C:\Users\Admin\AppData\Local\Temp\cc60rs88.o C:\Users\Admin\AppData\Local\Temp\ccHloCCy.s
GNU assembler version 2.24 (mingw32) using BFD version (GNU Binutils) 2.24
COMPILER_PATH=C:/TDM-GCC-32/bin/../libexec/gcc/mingw32/5.1.0/;C:/TDM-GCC-32/bin/../libexec/gcc/;C:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/../../../../mingw32/bin/
LIBRARY_PATH=C:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/;C:/TDM-GCC-32/bin/../lib/gcc/;C:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/../../../../mingw32/lib/;C:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/../../../
COLLECT_GCC_OPTIONS='-v' '-mtune=generic' '-march=pentiumpro'
 C:/TDM-GCC-32/bin/../libexec/gcc/mingw32/5.1.0/collect2.exe -plugin C:/TDM-GCC-32/bin/../libexec/gcc/mingw32/5.1.0/liblto_plugin-0.dll -plugin-opt=C:/TDM-GCC-32/bin/../libexec/gcc/mingw32/5.1.0/lto-wrapper.exe -plugin-opt=-fresolution=C:\Users\Admin\AppData\Local\Temp\ccnWR22L.res -plugin-opt=-pass-through=-lmingw32 -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lmoldname -plugin-opt=-pass-through=-lmingwex -plugin-opt=-pass-through=-lmsvcrt -plugin-opt=-pass-through=-lpthread -plugin-opt=-pass-through=-ladvapi32 -plugin-opt=-pass-through=-lshell32 -plugin-opt=-pass-through=-luser32 -plugin-opt=-pass-through=-lkernel32 -plugin-opt=-pass-through=-lmingw32 -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lmoldname -plugin-opt=-pass-through=-lmingwex -plugin-opt=-pass-through=-lmsvcrt --exclude-libs=libpthread.a -Bdynamic C:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/../../../crt2.o C:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/crtbegin.o -LC:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0 -LC:/TDM-GCC-32/bin/../lib/gcc -LC:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/../../../../mingw32/lib -LC:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/../../.. C:\Users\Admin\AppData\Local\Temp\cc60rs88.o -Bstatic -lstdc++ -Bdynamic -lmingw32 -lgcc -lmoldname -lmingwex -lmsvcrt -lpthread -ladvapi32 -lshell32 -luser32 -lkernel32 -lmingw32 -lgcc -lmoldname -lmingwex -lmsvcrt C:/TDM-GCC-32/bin/../lib/gcc/mingw32/5.1.0/crtend.o`

Is there any known workaround for getting a successful compilation?

Thanks in advance.
Regards
Alberto

RBEI-busmaster, 2016-08-11:
BusMaster v3.0.0 supports 'MinGW TDM1 GCC v4.8.1'.
Use 'Download MinGW' option and ‘User Manual’ option from the main page 'http://rbei-etas.github.io/busmaster/' to download and install the same.

Please refer section ‘Additional Installation->MinGW Installation using TDM-GCC Installer’ in the user manual for the installation steps.

roblamoreaux-swri, 2016-08-21:
I too cannot get this to work and I reinstalled as per the instructions so it has 4.8.1 (stable). Here is a log from the makefile Busmaster created:

C:\Users\Customer\Documents\AAProjects\03-22075\simulate>mingw32-make --makefile ECUmake
mingw32-g++ -include"C:\ProgramData\BUSMASTER\General\DBCache\F546.h"  -Wall -c -v -I"C:\Program Files (x86)\BUSMASTER_v3.0.0\SimulatedSystems\Include" C:/Users/Customer/DOCUME~1/AAPROJ~1/03-220~1/simulate/ECU.cpp -o ECU.o
Using built-in specs.
COLLECT_GCC=mingw32-g++
Target: mingw32
Configured with: ../gcc-4.8.1/configure --prefix=/mingw --host=mingw32 --build=mingw32 --without-pic --enable-shared --enable-static --with-gnu-ld --enable-lto --enable-libssp --disable-multilib --enable-languages=c,c++,fortran,objc,obj-c++,ada --disable-sjlj-exceptions --with-dwarf2 --disable-win32-registry --enable-libstdcxx-debug --enable-version-specific-runtime-libs --with-gmp=/usr/src/pkg/gmp-5.1.2-1-mingw32-src/bld --with-mpc=/usr/src/pkg/mpc-1.0.1-1-mingw32-src/bld --with-mpfr= --with-system-zlib --with-gnu-as --enable-decimal-float=yes --enable-libgomp --enable-threads --with-libiconv-prefix=/mingw32 --with-libintl-prefix=/mingw --disable-bootstrap LDFLAGS=-s CFLAGS=-D_USE_32BIT_TIME_T
Thread model: win32
gcc version 4.8.1 (GCC)
COLLECT_GCC_OPTIONS='-include' 'C:\ProgramData\BUSMASTER\General\DBCache\F546.h' '-Wall' '-c' '-v' '-I' 'C:\Program Files (x86)\BUSMASTER_v3.0.0\SimulatedSystems\Include' '-o' 'ECU.o' '-shared-libgcc' '-mtune=generic' '-march=pentiumpro'
 c:/mingw32/bin/../libexec/gcc/mingw32/4.8.1/cc1plus.exe -quiet -v -I C:\Program Files (x86)\BUSMASTER_v3.0.0\SimulatedSystems\Include -iprefix c:\mingw32\bin\../lib/gcc/mingw32/4.8.1/ -include C:\ProgramData\BUSMASTER\General\DBCache\F546.h C:/Users/Customer/DOCUME~1/AAPROJ~1/03-220~1/simulate/ECU.cpp -quiet -dumpbase ECU.cpp -mtune=generic -march=pentiumpro -auxbase-strip ECU.o -Wall -version -o C:\Users\Customer\AppData\Local\Temp\ccnBcQ77.s
GNU C++ (GCC) version 4.8.1 (mingw32)
        compiled by GNU C version 4.8.1, GMP version 5.1.2, MPFR version 3.1.2, MPC version 1.0.1
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "c:\mingw32\bin\../lib/gcc/mingw32/4.8.1/../../../../mingw32/include"
ignoring duplicate directory "c:/mingw32/lib/gcc/../../lib/gcc/mingw32/4.8.1/include/c++"
ignoring duplicate directory "c:/mingw32/lib/gcc/../../lib/gcc/mingw32/4.8.1/include/c++/mingw32"
ignoring duplicate directory "c:/mingw32/lib/gcc/../../lib/gcc/mingw32/4.8.1/include/c++/backward"
ignoring duplicate directory "c:/mingw32/lib/gcc/../../lib/gcc/mingw32/4.8.1/include"
ignoring duplicate directory "c:/mingw32/lib/gcc/../../lib/gcc/mingw32/4.8.1/../../../../include"
ignoring nonexistent directory "/mingw/include"
ignoring duplicate directory "c:/mingw32/lib/gcc/../../lib/gcc/mingw32/4.8.1/include-fixed"
ignoring nonexistent directory "c:/mingw32/lib/gcc/../../lib/gcc/mingw32/4.8.1/../../../../mingw32/include"
ignoring nonexistent directory "/mingw/include"
#include "..." search starts here:
#include <...> search starts here:
 C:\Program Files (x86)\BUSMASTER_v3.0.0\SimulatedSystems\include
 c:\mingw32\bin\../lib/gcc/mingw32/4.8.1/include/c++
 c:\mingw32\bin\../lib/gcc/mingw32/4.8.1/include/c++/mingw32
 c:\mingw32\bin\../lib/gcc/mingw32/4.8.1/include/c++/backward
 c:\mingw32\bin\../lib/gcc/mingw32/4.8.1/include
 c:\mingw32\bin\../lib/gcc/mingw32/4.8.1/../../../../include
 c:\mingw32\bin\../lib/gcc/mingw32/4.8.1/include-fixed
End of search list.
GNU C++ (GCC) version 4.8.1 (mingw32)
        compiled by GNU C version 4.8.1, GMP version 5.1.2, MPFR version 3.1.2, MPC version 1.0.1
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: 1ebc2a6f92fbd3aadc367a20a63fdf9f
COLLECT_GCC_OPTIONS='-include' 'C:\ProgramData\BUSMASTER\General\DBCache\F546.h' '-Wall' '-c' '-v' '-I' 'C:\Program Files (x86)\BUSMASTER_v3.0.0\SimulatedSystems\Include' '-o' 'ECU.o' '-shared-libgcc' '-mtune=generic' '-march=pentiumpro'
 c:/mingw32/bin/../lib/gcc/mingw32/4.8.1/../../../../mingw32/bin/as.exe -v -I C:\Program Files (x86)\BUSMASTER_v3.0.0\SimulatedSystems\Include -o ECU.o C:\Users\Customer\AppData\Local\Temp\ccnBcQ77.s
GNU assembler version 2.24 (mingw32) using BFD version (GNU Binutils) 2.24
COMPILER_PATH=c:/mingw32/bin/../libexec/gcc/mingw32/4.8.1/;c:/mingw32/bin/../libexec/gcc/;c:/mingw32/bin/../lib/gcc/mingw32/4.8.1/../../../../mingw32/bin/
LIBRARY_PATH=c:/mingw32/bin/../lib/gcc/mingw32/4.8.1/;c:/mingw32/bin/../lib/gcc/;c:/mingw32/bin/../lib/gcc/mingw32/4.8.1/../../../../mingw32/lib/;c:/mingw32/bin/../lib/gcc/mingw32/4.8.1/../../../
COLLECT_GCC_OPTIONS='-include' 'C:\ProgramData\BUSMASTER\General\DBCache\F546.h' '-Wall' '-c' '-v' '-I' 'C:\Program Files (x86)\BUSMASTER_v3.0.0\SimulatedSystems\Include' '-o' 'ECU.o' '-shared-libgcc' '-mtune=generic' '-march=pentiumpro'
mingw32-g++ -shared -Wl,--dll  ECU.o "C:\ProgramData\BUSMASTER\General\DBCache\F546_gcc4.8.1.obj"  -L"C:\Program Files (x86)\BUSMASTER_v3.0.0\SimulatedSystems\OBJ" -lWrapper_CAN -o ECU.dll
C:\Program Files (x86)\BUSMASTER_v3.0.0\SimulatedSystems\OBJ/libWrapper_CAN.a(Wrapper_Common.o):Wrapper_Common.cpp:(.text+0xcf): undefined reference to `__gxx_personality_sj0'
C:\Program Files (x86)\BUSMASTER_v3.0.0\SimulatedSystems\OBJ/libWrapper_CAN.a(Wrapper_Common.o):Wrapper_Common.cpp:(.text+0xfd): undefined reference to `_Unwind_SjLj_Register'
C:\Program Files (x86)\BUSMASTER_v3.0.0\SimulatedSystems\OBJ/libWrapper_CAN.a(Wrapper_Common.o):Wrapper_Common.cpp:(.text+0x178): undefined reference to `_Unwind_SjLj_Resume'
C:\Program Files (x86)\BUSMASTER_v3.0.0\SimulatedSystems\OBJ/libWrapper_CAN.a(Wrapper_Common.o):Wrapper_Common.cpp:(.text+0x186): undefined reference to `_Unwind_SjLj_Unregister'
collect2.exe: error: ld returned 1 exit status
ECUmake:20: recipe for target 'ECU.dll' failed
mingw32-make: *** [ECU.dll] Error 1

C:\Users\Customer\Documents\AAProjects\03-22075\simulate>

roblamoreaux-swri, 2016-08-21:
I got it to work by using the recommended TDM installation which installs version 5.1.0. The help file and documentation needs to be updated to not say use the stable installation that installs 4.8.1, but to install 5.1.0.

alberto77, 2016-08-22:
Hi  @RBEI-busmaster,
I downloaded the installer form the site you had linked. I installed a new version of TDM following the instructions reported in help file: I installed default version as mentioned. I got the same error: undefined reference to Unwind_Resume.

alberto77, 2016-08-22:
Moreover I have another error raised by ld.exe: bad reloc addresds 0x120 in section '.text$_ZN14InternalCANdbf5Frame...

alberto77, 2016-08-22:
I solve the problem creating a completely new node file. Now I get a successful compilation. Thank you very much for your help.

closed 2016-08-22 by alberto77