rbei-etas/busmaster #976: GCCDLLMakeTemplate_CAN not found

ones1, 2016-08-29:
Hello BM community,

when I load my BM-config-file with VS, I get this message in the output window:
"C:\WINDOWS\system32\GCCDLLMakeTemplate_CAN was not found"

but when I use BM alone to load the config-file it works as it should.

does someone knows about this problem?
I already copied the file to my system32 folder without result.

I would be grateful about any hint!

RBEI-busmaster, 2016-08-30:
Hi,
Please use the latest BusMaster v3.0.0. 
Please Copy the below folders/files to BUSMASTER\BIN\Debug or BUSMASTER\BIN\Release based on the configuration type(Debug/Release) used
1. BUSMASTER\Application\SimulatedSystems
2. BUSMASTER\Application\GCCDLLMakeTemplate

ones1, 2016-09-02:
I'll try it out! 
Thank you!