rbei-etas/busmaster #980: CAN dbc not converted to dbf

dinesh2501, 2016-09-19:
Hello,
I converted my CAN dbc to dbf using BUSMASTER converter tool.
When I assigned this dbf to BUSMASTER,it shows no message and signal in the busmaster. Eventhough the  dbf file is no blank.
please help.

RBEI-busmaster, 2016-09-26:
To analyze the issue, please provide the below details:
1. BusMaster Version used 
2. Send .dbc file used to BUSMASTER@in.bosch.com

whitefalk, 2019-01-08:
Hi,
This may be the same as #838
I had a similar issue, and I opened the DBC in CANdb++ editor, and saved it, and then it worked.

This is not the solution though. There is some kind of name abbreviation scheme in dbc that is not considered by the converter tool. There is a list of all message names but later a trunkated name appears, sometimes with the ending _0000 and _0001 etc.. Sometime the name is cut.

So CANdb++ editor resolved this, when I saved it again it resolved all these, and the full name is used now. My 800 kB dbc grew to about 900kB.

I can't currently attach an example, because it was from some proprietary project.

jogo-, 2019-02-07:
@whitefalk just have a look at the PR #1184