rbei-etas/busmaster #982: Database editor: how do I 'wildcard' message IDs to accept from all Source Addresses, Priorities etc.?

tomgreenflir, 2016-09-22:
I am looking at BM to see if it can help us with NMEA2000 field diagnosis. Perhaps I'm being dense but I cannot find out how I can create Database entries where Message ID can be 'wild-carded' to accept the same message from any Source Address or with any Priority. Essentially I think I need a mask, like in issue 414.

Example: I receive NMEA2000 PGN 127245 (0x1F10D) from address 0xCD with priority 2 (0x9F10DCD.) If I write my Database entry Message ID=0x9F10DCD, the message is recognised as Rudder type in the Message window, but this will only work if the message's Source Address (LS byte) is 0xCD and Priority=2. I don't want to have to write separate DB entries for all possible Source Addresses(!). Writing the DB MessageID=0x1F10D00, 0x1F10DFF, 0xFFF10DFF etc. all result in the message type not being recognised in traffic in the Message Window.

How can I write a DB entry so that the first 7 and last 8 bits of the Message ID are don't-care values?

Other than that, BM looks really powerful and I would love to be able to use it more.

alesiem, 2018-02-28:
Did anything happen with masking message IDs? 
I'm facing the exact same problem as described above!

tomgreenflir, 2018-04-26:
Following on from my above, I found that I could achieve the above using the J1939 Database Editor instead of raw CANbus editor; my subsequent problem was with multi-frame messages. I haven't yet worked out how to get BusMaster to concatenate the data fields from subsequent frames in the same multi-frame message.