rbei-etas/busmaster #986: busmaster can‘t run on win10

weiwei803, 2016-10-07:
OS:win10 enterprise (10240)   chinese (simplified)
System type:x64
software:Visual Studio 2013 Redistributable (x86),TDM-GCC 4.8.1(32-bit)

After the clicking action finished,BUSMASTER start up and show the splash,  after that, the program freezes,and can't be closed .

I  tried to adjust the compatibility mode,but the issuse was still there.

[image]

ETAS-Nithin, 2016-10-25:
Could you please share the log files to busmaster@in.bosch.com ? We have performed partial testing on Win 10 Pro x64 (Eng) and the freezing issue was not observed.

weiwei803, 2016-11-03:
@ETAS-Nithin Thanks for your reply.Today,I uninstall busmaster,and reinstall it,the problem disappear,busmaster run well.