rbei-etas/busmaster #987: Matlab via Busmaster

busmaster3, 2016-10-11:
Hi,

i'm just doing my Thesis ans i would like to let busmaster control over matlab,
ans send the can Messages through kvaser.

It is actually possible through COM API
if so can you send me a sample functions.

best regards 
youssef

busmaster3, 2016-10-12:
I explain my question more precisely

I would like to know whether there is the possibility that I can control ETAS busmaster over MATLAB

That is, from starting the ETAS bus master to sending the CAN messages.

I have opened up now through COM API, the program Busmaster

And now I would like to continue with:

Select Device

CAN Connect

Send CAN_news

etc....
Maybe you have a tutorial or example functions.

Can you please send me the tips or give some hints how I can implement MATLAB access to ETAS bus master

thank you

RBEI-busmaster, 2016-10-19:
Hello,
Please refer help section 'CAN->COM interface' for COM API documentation.
Sample COM client 'VC++COM_Client.zip' is available under BusMaster installation folder 'Examples\COM'.