rbei-etas/busmaster #991: "unhandled exception in BUSMASTER."when touch the connect/disconnect botton

Wenji2016Jim, 2016-11-17:
[image]

pls. help to solve. thank you in advance.

ETAS-Nithin, 2016-11-23:
@Wenji2016Jim could you please provide further details on the OS, Hardware connected (if any) etc ? Could you please share the log file to busmaster@in.bosch.com ?

Wenji2016Jim, 2016-12-01:
ok i will wirte email about the detailed information. thank you.

Wenji2016Jim, 2016-12-19:
@ETAS-Nithin, do you have sloved this issus?  i need the help.

Wenji2016Jim, 2016-12-20:
@ETAS-Nithin @ETAS-COM 
i had tried to connect the CAN first and then power on the vehicle , the Busmaster can work normally;
but if i power on vehicle first, and then connect the CAN, it also shows the up issue.  
also i tried to install version 2.6.0/2.6.4/3.0/3.1 ; all of them are the same.
pls. help to sovle it ASAP. I need your help.

cuietviper, 2017-02-21:
This issue is dude the following reason
Unhandled exception at 0x0D98EF9B (CAN_ETAS_BOA_2.dll) in BUSMASTER.exe: 0xC0000094: Integer division by zero.
Please look into this 
@ETAS-COM @ETAS-Nithin

wangqf74, 2017-09-20:
I have the same problem.
Could you tell me the solution?