rbei-etas/busmaster #994: Missing mfc120.dll in BUSMASTER v3.0.0

piyushk93, 2016-11-30:
System configuration:
Windows 7.
system type: 64 bit os
[image]

While starting BUSMASTER v3.0.0 it gives error that the program can't start because mfc120.dll is missing from your computer, try reinstalling the program to fix this problem.
after re-installation the same error occurs.

RBEI-busmaster, 2016-11-30:
Hi,

BUSMASTER v3.0.0 requires Visual Studio 2013 Redistributable installation.
For installation, please refer the link https://github.com/rbei-etas/busmaster/wiki/Visual-Studio-2013-Redistributable.

Closing the issue, As it is similar to #950. #950 will be used for tracking.

closed 2016-11-30 by RBEI-busmaster

piyushk93, 2016-11-30:
Thanks for your solution, it's working.