rbei-etas/busmaster #998: Data is not received completely

piyushk93, 2016-12-01:
int j;
    int i;
     int RxBuff[8];
     for(i=0;i<8;i++)
     {
	     RxBuff[i]= RxMsg.byteAt(i); //RxMsg.data[i];
     }
     for(j=0;j<8;j++)
     {
		Trace("%d",RxBuff[j]);
     }

I am trying to collect the received message data in buffer but only 4 bytes are loaded into buffer every time, Can you tell me how to load complete received message data into buffer

piyushk93, 2016-12-02:
int j;
int i;
 int RxBuff[8];
 for(i=0;i<8;i++)
 {
     RxBuff[i]= RxMsg.byteAt(i); //RxMsg.data[i];
 }
 for(j=0;j<8;j++)
 {
	Trace("%d",RxBuff[j]);
 }
 for(j=0;j<8;j++)
 {
	Trace("%d",RxMsg.byteAt(j));
 }
In j1939 node simulation, i was tried to collect the received message data in buffer but only 4 bytes were loaded into buffer every time, also i was tried to print the  RxMsg.byteAt(j) in trace window but only 4 LSB bytes were printed correctly. can you kindly tell me the solution for this problem as early as possible.

RBEI-busmaster, 2017-03-31:
Fixed with v3.2.0. Please download and use the latest version

closed 2017-03-31 by RBEI-busmaster