rbei-etas/busmaster #1008: Issue with CAN Diagnostic Windows

longtranchau, 2016-12-06:
I have an issue with CAN Diagnostic Windows. It seems that Busmaster 3.x.x (both 3.0.0 and 3.1.0) can't handle multi-frame response from ECU. After ECU sent first frame, Busmaster didn't send flow control frame and the communication was cancelled.

[image]

If I use Busmaster 2.6.4, it's ok. Busmaster 2.6.4 can handle multi frame response from ECU.

[image]

Regards,
Long Tran

ddeussen, 2016-12-12:
I have basically reported the same issue https://github.com/rbei-etas/busmaster/issues/995 but have not received any feedback for 12 days now.

wenluoma, 2016-12-23:
could you tell me where i can download BUSMASTER 2.6.4???

RBEI-busmaster, 2017-03-31:
Fixed with v3.2.0. Please download and use the latest version

closed 2017-03-31 by RBEI-busmaster