rbei-etas/busmaster #1016: No Channel Configuration with IXXAT VCI and Busmaster 3.1.0

FAMOSTester, 2016-12-12:
Hello together,

actually i try the new version of Busmaster (3.1.0) and want to connect my IXXAT USB-to-CAN compact module (VCI 3.5.4.4543 / VCI 2.20.0.855). 
The driver can be selected and the module be started. But i can't configure the module. If i press the "Channel Configuration" button... nothing happens.
If i try the older version i can configure my module (baudrate etc.). The Vector Driver runs...
Any ideas?

nostar, 2016-12-14:
See #1017 

This has been labelled a support issue, but is in fact a bug in 3.1.0.

FAMOSTester, 2016-12-15:
Good to know. Then i will wait for an update. Thanks for the reply!

RBEI-busmaster, 2016-12-19:
Duplicate to #1017. #1017 will be used for tracking. Hence, closing the issue.

closed 2016-12-19 by RBEI-busmaster

RBEI-busmaster, 2016-12-19:
@FAMOSTester,  We would like to get your support in testing the fix with IXXAT hardware. Could you please send your contact details to BUSMASTER@in.bosch.com.

RBEI-busmaster, 2017-02-10:
Hi,
Please install the patch from the below location and let us know if it addresses your issue.
https://github.com/rbei-etas/busmaster-transient/blob/master/BUSMASTER_Patch_v3.1.0.exe 

Please write to busmaster@in.bsoch.com if any issues are found.
Based on the feedback, the patch will be made available as part of next version.