rbei-etas/busmaster #1017: No ability to change hardware configuration on a single channel device

nostar, 2016-12-14:
In 3.1 the 'Channel Selection' was renamed to 'Channel Configuration', and what used to be 'Channel Configuration' is now an advanced button on the channel selection dialog.  The problem is that the Channel Configuration button is not active for a single channel device, so now there is no way to change configuration settings, such as baud rate, on a single channel device.  This is true for Peak and Vector hardware, and IXXAT hardware according to #1016

RBEI-busmaster, 2017-02-10:
Hi,
Please install the patch from the below location and let us know if it addresses your issue.
https://github.com/rbei-etas/busmaster-transient/blob/master/BUSMASTER_Patch_v3.1.0.exe

Please write to busmaster@in.bsoch.com if any issues are found.
Based on the feedback, the patch will be made available as part of next version.

RBEI-busmaster, 2017-03-31:
Fixed with v3.2.0. Please download and use the latest version

closed 2017-03-31 by RBEI-busmaster