rbei-etas/busmaster #1027: Busmaster not setting bus speed correctly

bomber1973, 2016-12-19:
Busmaster v3.1.0 & Kvaser interface;

Changing speed from 500kbps to 1Mbps has no effect.  The only way it seems to get it to talk to a 1Mbps network is to have another tool that uses the kvaser API to configure it and then use busmaster.

RBEI-busmaster, 2016-12-23:
v3.1.0 has Channel configuration issue with Single channel hardware. 
Is the hardware used has only one Channel?

bomber1973, 2016-12-24:
The adapter is actually a 4 channel device, although we get the same problem configuring just one channel or all 4.

RBEI-busmaster, 2017-01-23:
We would like to get your support in testing the fix with Kvaser hardware. Could you please send your contact details to BUSMASTER@in.bosch.com.

RBEI-busmaster, 2017-02-10:
Hi,
Please install the patch from the below location and let us know if it addresses your issue.
https://github.com/rbei-etas/busmaster-transient/blob/master/BUSMASTER_Patch_v3.1.0.exe

Please write to busmaster@in.bsoch.com if any issues are found.
Based on the feedback, the patch will be made available as part of next version.

Jeroen6, 2017-04-03:
v3.2.0 seems to fix it. 
But BTR0 isn't calculated correctly on the first try.