rbei-etas/busmaster #1031: errors when run with debug mode

wenluoma, 2016-12-26:
when i run with debug mode, it pops up a message window like the following: 
so please tell me the reason? i tried to change a big memory but it still have this problem.

[image]
[image]

RBEI-busmaster, 2017-01-03:
Hi,
Please execute the build.bat located at \busmaster\Sources\ which builds in release mode and let us know if you are able to launch BUSMASTER.
We compiled the following solutions in the below order using Visual Studio 2013 in Debug mode and it worked fine without any issues.

1. busmaster\Sources\Kernel\BusmasterKernel.sln
2. busmaster\Sources\BUSMASTER\BUSMASTER.sln

wenluoma, 2017-01-04:
Hello,

first, i tested the BUSMASTER souce code with release mode, although there is still some issues about the build enviroment, i can fix these problems and the BUSMASTER main interface could be launched,and the Diagnostics function seems without any problems. so it seems work well now.
then i try to use Diagnostics function with Debug mode. however. when i click "Diagnostics" --> "Settings", there is a error, please see the attach. is there anything i missed?

wenluo

garnett33, 2017-02-09:
Hi, 

I tried to execute the build.bat as you proposed but it is still not working in both modes debug and release. 
Moreover, when I run it in debug mode, I have the same problem than wenluoma : "Insufficient Memory Available" message pops up. I think it could be a Visual Studio configuration issue so could you please tell me what parameters you set before you compile and run the project ? 

Thank you in advanced, 

Maxime

RBEI-busmaster, 2017-03-13:
@Maxime, we are unable to reproduce the issue. Could you please send your contact details to Busmaster@in.bosch.com. We can discuss further.

closed 2017-04-24 by RBEI-busmaster