rbei-etas/busmaster #1034: BTR calculation error

PPSXIN, 2017-01-05:
On the preview version like v3.0.0, when the Baud Rate is set to 125000 bps, the BTR0 will be 0x47 and BRP will be 8, it's ok to work.
but on the version v3.1.0, if the Baud rate is set to 125000 bps, by default the BTR0 will be 0x41 and BRP will be 2, it can't be connected with CAN bus. If these parameters are changed manually as the preview version, it's ok.
v3.1.0 screen shoot:
[image]
v3.0.0 
[image]

RBEI-busmaster, 2017-01-23:
We would like to get your support in testing the fix. Could you please send your contact details to BUSMASTER@in.bosch.com.

RBEI-busmaster, 2017-02-10:
Hi,
Please install the patch from the below location and let us know if it addresses your issue.
https://github.com/rbei-etas/busmaster-transient/blob/master/BUSMASTER_Patch_v3.1.0.exe

Please write to busmaster@in.bsoch.com if any issues are found.
Based on the feedback, the patch will be made available as part of next version.

geier99, 2017-02-11:
Hi,
my BTR calculation issues are solved. ( for example I wasn't  able to set the baudrate to 33333 bps).

RBEI-busmaster, 2017-02-13:
@geier99, Thanks for the feedback. Could please let us know the Operating System used.

geier99, 2017-02-13:
Hi,
I have used: Windows 8.1 Pro   ( with PCAN-USB interface)
And today it solved also my issues on a Windows 7 Pro system.  ( with  PCAN-USB Pro interface)

PPSXIN, 2017-02-15:
@RBEI-busmaster 
Hi,
Sorry for the late feedback.

The installation failed when I tried to install the patch file:
[image]

PPSXIN, 2017-02-15:
BTW, the OS is win7 X64.

RBEI-busmaster, 2017-03-31:
Fixed with v3.2.0. Please download and use the latest version

closed 2017-03-31 by RBEI-busmaster