rbei-etas/busmaster #1036: Peak - USB channel configuration  V3.1.0

serci09, 2017-01-14:
Hello,
          Once I connect to The PEAK USB driver, I cannot configure channel. It is currently set at 500 kbaud and I need to change it to 250 Kbaud. Is there a different way to do it on V 3.1.0.    Version V3.0.0 works without any issues

RBEI-busmaster, 2017-01-16:
We would like to get your support in testing the fix with PEAK hardware. Could you please send your contact details to BUSMASTER@in.bosch.com.

serci09, 2017-01-21:
Email sent

airmaciej, 2017-02-03:
Hello,
I have also an issue with Peak USB in V3.1.0 . After configuring channels (I cannot see the channel numbers) I receive only error frames. Maybe extended CAN ID (29bit) is used as default? In V2.6.0 the Peak USB is working correctly.

RBEI-busmaster, 2017-02-10:
Hi,
Please install the patch from the below location and let us know if it addresses your issue.
https://github.com/rbei-etas/busmaster-transient/blob/master/BUSMASTER_Patch_v3.1.0.exe

Please write to busmaster@in.bsoch.com if any issues are found.
Based on the feedback, the patch will be made available as part of next version.

geier99, 2017-02-11:
Hi,
I  had the same issue and on my side and the issue has being solved. 

 

I have a PCAN-USB dongle and a PCAN-PCI card in my computer. 
Now I can configure the baudrate settings for the PCAN-USB. 

Thank you!

   Andreas

 

https://github.com/rbei-etas/busmaster-transient/blob/master/BUSMASTER_Patch_v3.1.0.exe

yegorich, 2017-02-20:
@RBEI-busmaster could you please push this commit.

By the way could commits be made more atomically, so that one would have a chance to use `git blame` or `git bisect` at all. One commit with 1300 changed files is very hard to inspect.