rbei-etas/busmaster #1038: Format Converter DBF to DBC error v3.1.0

wangchengze, 2017-01-18:
dbf:
signal Detail:
Name:PresentHour;
type:unsigned int, Byte Index:3, Start Bit:6;
Length:5 Bits, Min Val:0x0, Max Val:0x1F;
Offset:0, Factor:1, Unit: ;
Byte Order:Motorola(Big-Endian);
in dbf file:
START_SIGNALS] PresentHour,5,4,6,U,31,0,0,0.000000,1.000000,,

busmaster v3.1.0
dbf to dbc
in dbc file:
SG_ PresentHour : 34|5@0+ (1,0) [0|31] "" Vector__XXX

the bits of this signal should be:18,17,16,31,30( MSB <-- LSB )

but in dbc, the bits are not right.

wangchengze, 2017-01-18:
[image]

RBEI-busmaster, 2017-01-20:
Could you please share the sample files to "busmaster@in.bosch.com"