rbei-etas/busmaster #1042: Crash when selecting IntrepidCS NeoVI

SierraTangoR2, 2017-01-31:
Precondition: IntrepidCS NeoVI Fire Red w/ ICS vehicle spy and drivers already installed (tested+confirmed)
Action: Open Busmaster 3.1.0, select CAN--> Driver Selection --> IntrepidCS NeoVI.
Observed Behaviour: Application crashes, log files saved (and attached)
Expected Behaviour: Application does not crash, selects driver for ICS tool, and works.

[BUSMASTER_CRASH_LOG.zip](https://github.com/rbei-etas/busmaster/files/741234/BUSMASTER_CRASH_LOG.zip)

RBEI-busmaster, 2017-02-07:
BusMaster uses icsneo40.dll to communicate with IntrepidCS hardware.
Please provide the version of ‘icsneo40.dll’ installed in your PC, which will be available at either of these locations ‘C:\Windows\System32’ or ‘C:\Windows\SysWOW64’
 
To get the version, please right click on the .dll, select ‘Properties’ and go to ‘Details’ tab as shown below.

SierraTangoR2, 2017-02-08:
The version listed in the DLL is 3.7.0.114.

intilinux, 2017-02-24:
I have the same issue with icsneo40.dll version 3.8.0.105.

RBEI-busmaster, 2017-10-16:
Please check with v3.2.2 beta uploaded in the transient repository

SierraTangoR2, 2017-10-25:
Fantastic! I no longer have access to the NeoVi but do have access to ValueCAN3 (which exhibited the same crash) - I can confirm the application does not crash.

Of interest though, the reported serial number is 50000 out. My units are SN: #87XXX and Busmaster reports the SN as being 37XXX. ie. the serial number is correct in the final 4 digits, but the first digit is off. 

Functionally, I am unsure this will affect anything, but  just a heads up.

Cheers,

RBEI-busmaster, 2017-10-30:
Issue is fixed with v3.2.2

closed 2017-10-30 by RBEI-busmaster