rbei-etas/busmaster #1047: Support for LAWICEL type hardware

riodda, 2017-02-26:
Would be nice to support low cost serial protocol hw like LAWICEL or CAN-TIN, there is full support in linux for SocketCAN.
I can gift a developer one board for thesupport, i have one but my coding skills are not good enough to add it in the support.

igittigitt, 2017-04-09:
I would be pleased to see this also. Why is there only support for this expensive commercial devices? Hey, this is opensource here!

ETAS-Nithin, 2017-04-12:
@riodda could you please write to us at busmaster@in.bosch.com ? We could then discuss on the integration part and implementation timeline.

yegorich, 2017-04-12:
It would be also good to be able to specify not only serial port but an IP address and port too. There are devices on the market, that accept LAWICEL commands (slcan) over TCP.

koenlek, 2017-06-15:
Any updates to this? @riodda: did contacting busmaster@in.bosch.com  result in any news?

eddieddi, 2018-03-20:
News on this ? many guys have this interface or lawicel compatible interfaces...
pleas add this hardware support.

dremwilly, 2018-06-02:
@igittigitt  - yes its opensource so Implement it if you need it - and share it :-)