rbei-etas/busmaster #1050: Busmaster 3.1.0 failed to recognise IXXAT CAN-to-USB V2 under VCI driver V4

natthapolvanasrivilai, 2017-03-23:
Busmaster 3.1.0 can only connect to the IXXAT USB-to-CAN V2 when VCI V3 is installed.
It shows "driver selection failed" when use with VCI V4.

link to VCI V4: https://www.ixxat.com/support/file-and-documents-download/drivers/vci-v4-driver
link to VCI V3: https://www.ixxat.com/support/file-and-documents-download/drivers/vci-v3-driver-download

jbiefm, 2017-08-01:
I have the same issue.
For compatibility issues with other code, I would prefer to use VCI V4.
Is there an idea whether and/or when this will be addressed, i.e. VCI V4 will be supported/recognized?

bodegaard, 2017-11-17:
My new Windows 10 machine will refuse to boot if there's unsigned drivers installed. Even getting an unsigned driver installed in the first place is a hastle. The problem reoccurs every time windows does an update and I've got to do a workaround every time to get the signature check re-disabled.

VCI V4 driver has a signed version, however VCI V3 does not. This means that I cannot use my USB-CAN Compact with busmaster without this work around.

polyene, 2018-01-30:
I'm also having this issue. My new Windows 10 PC with secure boot refuses to boot with the VCI3 driver installed. All other programs that I have that were designed for VCI3 seems to work fine with the VCI4 driver, so I don't know why BusMaster doesn't. 

This seems like a critical issue for anyone using the IXXAT interfaces, since you're sooner or later forced to upgrade to VCI4.

I'm also using Busmaster 3.2.2.

bodegaard, 2018-01-30:
I've found a work around for now! There's a specific version that works. see the faq.

"

> Q: Why the VCI V3.5.4.4543 does not run with the Windows 10?
> Please deactivate the Secure-Boot in the BIOS or use the "VCI V4.0.482.0 Attestation Signed" or use the VCI 3.5.2.4268.

"

[link](https://www.ixxat.com/support/faq-and-tips/driver-and-api/vci-driver-installation-faq)

polyene, 2018-01-31:
Thank you bodegaard, the workaround worked fine for me!

But I'm still hoping to see VCI4 support in the future to make the installation process easier.

castelaoguillermo, 2022-06-13:
Hello everyone!
I am using VCI version V4.0.1003.0 for windows 10 as V3 is not supported.
I have created the Windows enviroment variable 'IXXAT_VCI_SDK' with the path to the VCI header files "C:\Program Files\IxxatVCI\sdk\vci\inc" and i am having this error:
[image]

Does anyone know if it is related to the fact that I am using VCI version 4 instead of 3?
Thank you all!

az-eddinehaidavemsolutions, 2023-11-06:
Hi,
is there support for Ixxat simplyCAN?
thanks