rbei-etas/busmaster #1064: No Ribbon on Win7 (Japanese Workstation)

PTescAL, 2017-04-06:
In version 3.2.0, the ribbon is entirely missing on my Windows 7 machine. Also, the title bar's icon, which is supposed to be sitting on the left, is not showing.

I work on a Japanese Workstation and the installer seems to detect that since the application's title is changed. However I am not given a choice as to the locale I want to use.

I haven't tested earlier versions.

PTescAL, 2017-04-21:
After having deleted the folder JPN in the installation directory, the locale defaulted to English and the ribbon was visible again.

PTescAL, 2017-04-21:
I'll close this issue since my problem is resolved.

closed 2017-04-21 by PTescAL

aptj-sakurai, 2017-07-31:
Same issue remains at V3.2.1 and same workaround (removing the "JPN" folder) works.

Note that, this issue was once fixed before, but revived again. See:
https://github.com/rbei-etas/busmaster/issues/476
https://github.com/rbei-etas/busmaster/issues/629

ETAS-Nithin, 2017-08-07:
It was retained since we were still evaluating Japanese language support. We have closed this topic and from the subsequent version, the JPN folder will not be part of the release. English text will be displayed for all OS languages.

aptj-sakurai, 2017-08-07:
Thank you for your feedback. I suppose , "no support for Japanese language" is much better than "not working at Japanese environment at all". I also support your decision.

ETAS-Nithin, 2017-08-07:
you are welcome and thank you for supporting the decision. Over last few years, we had feedback from quite a few users of Japanese OS and other language users that they are comfortable working with English text version rather than having issues/bugs associated with dedicated language versions. 

When we integrated Windows 10 x64 ENG support, it was planned to gradually discontinue other language versions.

RBEI-busmaster, 2017-10-16:
Please check with v3.2.2 beta uploaded in the transient repository

aptj-sakurai, 2017-10-17:
Thank you for the beta release. I checked it on Win7 64bit (JPN) with tdm-gcc-4.8.1.exe.
Ribbon (with English texts) appears appropriately.

There're some issues

1) (major, or maybe this conversion feature is just not supported) Tab "Tools": when I opened "Format Converter", there's an error dialog "attempt to unsupported operation" (in Japanese). Also then the dialog "BUSMASTER Format Conversions" appeared, but always show same error dialog when
* try to switch to other tabs in the dialog "BUSMASTER Format Conversions"
* try to on/off the check box "save converted database" (in Japanese) in the tab "CAPL_2_CPP convertor"

2) (minor) some of texts are still displayed in Japanese texts, for example:
* Trace window title in Japanese
* Tab CAN, dialog "hardware selection" from Channel Configuration: "hardware" in Japanese
* Tab J1939: "Activate" button in Japanese
* BUSMASTER Format Conversions: mostly shown as 

Is there any option to switch to all English language version? Probably it will be more stable.

RBEI-busmaster, 2017-10-26:
Thanks for the feedback. An updated beta version is uploaded to transient repository. Could you please check if the above issues are addressed ?

aptj-sakurai, 2017-10-30:
I checked the newer beta on Win10 32bit (JPN) with tdm-gcc-4.8.1.exe.
All of the issues which I reported before (listed below) seems fixed well. All of them are shown in English without error dialog.
Thank you for your update!

1) (major, or maybe this conversion feature is just not supported) Tab "Tools": when I opened "Format Converter", there's an error dialog "attempt to unsupported operation" (in Japanese). Also then the dialog "BUSMASTER Format Conversions" appeared, but always show same error dialog when
  * try to switch to other tabs in the dialog "BUSMASTER Format Conversions"
  * try to on/off the check box "save converted database" (in Japanese) in the tab "CAPL_2_CPP convertor"

2) (minor) some of texts are still displayed in Japanese texts, for example:
  * Trace window title in Japanese
  * Tab CAN, dialog "hardware selection" from Channel Configuration: "hardware" in Japanese
  * Tab J1939: "Activate" button in Japanese
  * BUSMASTER Format Conversions: mostly shown in Japanese