rbei-etas/busmaster #1072: Rx extended trames aren't displayed using NSI hardware

MelissaA74, 2017-05-05:
Hello,

I can't receive extended type messages with a MuxyBox2 from NSI. 

Description of the procedure:
1.	I open Busmaster
2.	I set the driver selection on NSI CAN-API 
3.	I select the two first channels.
4.	I set the extended acceptance filter mode for the two channels in the filter option of the channel configuration window. 
5.	I create a cyclic extended frame in the transmit window.
6.	I connect the hardware interface.
7.	The transmitted frames are well displayed in the Message Window but not the received frames (I precise that I have loopbacked the Muxybox2 channels to use only one PC). 
8.	The transmitted frames are counted in the Bus Statistics but not the received frames.

I am currently using the latest version of Busmaster 3.2. And I know that the Rx extended messages were well displayed in the 3.0 version.

Thank you in advance. 

Best regards,

Mélissa

RBEI-busmaster, 2017-06-30:
Hello,
We assume the frames are sent in channel 1 and expected to receive in channel 2
Do you receive any error frames ?
What is the behavior with BusMaster v3.0.0 ?

MelissaA74, 2017-09-20:
Hello,

Thank for your answer. Yes indeed, the frames are sent from channel 1 and received  by channel 2. The behaviour can be observed from the v3.1.0, and the CAN Channel selection window enhancement that has been made.

I have finally resolved the issue. Because of the channel selection window enhancement, the call to nSetApplyConfiguration() function had been deleted in the CAN_NSI project.  I have also found some other bugs I will submit you soon in a pull request.

Mélissa

jogo-, 2018-01-14:
I don't see any waiting pull request for the above commit.
Let's hope these changes won't be lost/forgotten in next release...