rbei-etas/busmaster #1085: Any chance to run an old MHS Tiny I adapter on V3.x ?

mibcat, 2017-06-24:
Hi all!
I'm  not able to run an old MHS Tiny I Can adapter (firmware 1.21) with Busmaster V3.x (currently V3.2.1).
The "trick" during interface setup used in older Busmaster versions [issue 774](https://github.com/rbei-etas/busmaster/issues/774) does not work anymore.

Is it time to buy a new one or is there still a chance ?

Best regards,
Michael.

nostar, 2017-06-27:
Get TinyCan_431.exe and install latest drivers and firmware, then get the Busmaster bugfix zip file from the same page and overwrite the CAN_MHS.dll in your install directory with the one provided.

http://www.mhs-elektronik.de/index.php?module=download&action=list

mibcat, 2017-06-28:
Hi nostar,
thanks for the feedback.
With the bugfix **CAN_MHS.dll** file Busmaster throws immediately an error "Auflistung der Hardwareschnittstellen fehlgeschlagen" (hardware interface listing failed) as soon as I pick the driver **MHS Tiny-CAN** from the list. Also the channel configuration box is not launched as it is with the original dll.
???

nostar, 2017-06-30:
You need the latest drivers and firmware (firmware 1.65) from the TinyCan_431.exe for it to work.

mibcat, 2017-06-30:
Where do I find this firmware version ?
After installation of TinyCan_431.exe I've found a folder tiny_can/TCanProg containing the program tcanprog.exe.  But once started there is no way to choose a firmware. The program gives me the following output:

**Konfigurationsdatei "flash.ini" geladen
Datei Liste "files.ini" Datenbank geladen
Konfigurationsdatei "flash.ini" geladen
Connecting... Ok
Verbindung zur Hardware hergestellt
Informationen zur Hardware: Tiny-CAN I Hardware-Revision: 1.0
   Programm: Tiny-CAN I (Version: 1.210)
Programmdatei "Tiny-CAN1.mhs" geladen
Informationen zur Datei:
   Programm: Tiny-CAN I (Version: 1.210)
Blank-Check  ... FEHLER
Flash Speicher nicht geloescht
Flash Clear  ... Ok
Blank-Check  ... Ok
Flash Write  ... Ok
Aktion(en) erfolgreich beendet**

Obviously it takes the file **Tiny-CAN1.mhs** which is of the same version **1.210** currently flashed on the device.

BTW: the output of the program **TCanCheck.exe** shows as firmware version **1.21**

nostar, 2017-07-18:
My firmware before updating to 1.65 was 1.55, much newer than yours, and also didn't work with the bugfix DLL.  I would contact MHS for direction on how to update your firmware.