rbei-etas/busmaster #1088: Busmaster v3.2.0 unstable

karthikchittur, 2017-07-03:
Busmaster v3.2.0 is causing blue screen on my machine.
Machine : Zbook 15 with win7

Scenario 1: 
unplugging of  usb cable {re

scenario 2: 
while on battery and using busmaster go to sleep mode and wake up again connected to power supply

hint: i feel here some display driver might not be handled correctly.
laptop works in hybrid mode : on battery it uses internal intel display driver and on power supply it uses Nvidia driver.

ETAS-Nithin, 2017-07-31:
What USB-XX hardware was used with Busmaster when the crash happened ? Do you face the same issue with v3.2.1 as well ?

karthikchittur, 2017-07-31:
Hello,

We are using 581.3.
I have still not used the latest version. Will check this week.

Best regards,

C V Karthik

Engineering Solutions - Hybrid Power Train (RBEI/EHV3)
Robert Bosch Engineering and Business Solutions Private Limited - (CIN: U72400KA1997PTC023164) | 1st Flr, Campus 3A, EcoSpace
Varthur Hobli, Sarjapura | Bangalore(Bellandur)
Karnataka, 560 103 | INDIA | www.bosch-india-software.com<http://www.bosch-india-software.com>
Tel. +91 80 6783-4934 | Fax +91 80 6617-0711 | Karthik.CV@in.bosch.com<mailto:Karthik.CV@in.bosch.com>

Registered Office: Robert Bosch Engineering and Business Solutions Private Limited, 123, Industrial Layout,
Hosur Road, Koramangala, Bangalore - 560095, India
Managing Director: Mr. Vijay Ratnaparkhe,