rbei-etas/busmaster #1093: CAN Replay with Peak USB does not send any messages without loop

Dietrich87, 2017-07-26:
If I try to replay a CAN .log file with a PEAK USB interface, it does not send any message on the bus. Even if I connect some other logger to the CANBUS to check if there is something. If I choose the "simulation" it works fine, the Message Window shows the messages from the .log file.

I got it working using two PEAK USB interfaces. One is sending the messages, the other one is receiving. Even if I check the bus between the two interfaces with another device, all the messages from the .log file appear. With this configuratiuon it works fine, but I don't think it is intended wo work like this.

I was checking the documentation, but I still cannot find any reason for this behaviour.

RBEI-busmaster, 2017-07-31:
Is there any filter applied to the configuration in which messages are not appearing from the log ? Replay is independent of the hardware used.

Please send us the configuration and the log file to check the issue.

Dietrich87, 2017-07-31:
there is no filter applied. Using the "Simulation" instead of the PEAK Hardware shows all the messages.

[logfiles.zip](https://github.com/rbei-etas/busmaster/files/1186482/logfiles.zip)