rbei-etas/busmaster #1094: package release problems with busmaster 3.1.0

wenluoma, 2017-07-31:
i have download busmaster 3.1.0 and i have add some UDS function base on this version.
so i want to release this, and  package to installer. but i ahve meet some problems.

i use vs and new a project with type "InstallShield Limited Edition Project"
[image]

[image]

the "Application Files" part, i click"Add Files"button and add file "\Sources\BUSMASTER\BIN\Release\BUSMASTER.exe", then build and get a package, i can install the package but when i click the shortcut, there is a error 

[image]

looking forward your hep. thanks.

wenluo

ETAS-Nithin, 2017-07-31:
Hi wenluo, is there a specific reason for using 3.1.0 and not the newer releases for developing the UDS functionality ? 

Also, would you be interested to contribute the UDS implementation to the community ?

wenluoma, 2017-08-02:
there is no specific reason for using v3.1.0, because when i have download busmaster from github, the newer release(3.2.0 and later) is not available.
as contribute to the community, i am pleased to do this. i develop this tool for diagnostic test for our engineers.
if i can fix this problem, i think i will submit this to the community.
these UDS function include flash program the vcu, DTC,LID routine....
could you tell me how dou you release it, i have know that you have relaese v3.1.0 and v3.2.0, and i have used them. so could you give me some instructions for release it?
i am very appreciate for your help and i think continue to develop some functions based on v3.1.0 or newer version.

wenluo

ETAS-Nithin, 2017-08-07:
Thanks for the info. Sure, we could guide you with the process. Please write to us at busmaster@in.bosch.com and we could take it up.

wenluoma, 2017-08-08:
i have send email, i will not develop other more functions until this problem solved, so please help me and guide me. if possible, you can use teamviewer.