rbei-etas/busmaster #1098: Test executor only sends messages on simulated channel (not on IXXAT VCI)

itiserik, 2017-08-18:
BUSMASTER version 3.2.1
Steps to reproduce:
-Setup a BUSMASTER connected to a simulated channel
-Associate with a .dbf (CAN_Database.dbf)
-Open Test Automation Editor with (SampleTestSetupFile.xml)
-Verify that DataBase Path points to the same CAN_Database.dbf
-Open Test Automation Executor
-Add SampleTestSetupFile.xml
-Execute
-Verify in message window that Tx message is send out.
-Disconnect from channel
-Switch driver to IXXAT VCI (set baudrate etc)
-Connect to channel
-Open Test Automation Executor
-Execute
-No message is transmitted!

Transmission from Transmit Window is still possible.

Do I miss something here?
Can someone try to reproduce this on a development environment?

Thanks in advance!

RBEI-busmaster, 2017-08-22:
Is it happening only when transmitted with development environment ? 
How is it with the installation ? Could you please check and update ?

itiserik, 2017-08-22:
Yeah I found the issue:
At Sources/BUSMASTER/CAN_IXXAT_VCI/ClientList.cpp:123 I needed to add:
pdwClientID = GetClient(pacClientName)->dwClientID;

Can you integrate this?

RBEI-busmaster, 2017-09-07:
Yes, this will be integrated as part of the next open source version

RBEI-busmaster, 2017-10-16:
Please check with v3.2.2 beta uploaded in the transient repository

itiserik, 2017-10-30:
Seems to be solved now #1113 is bugging me.

RBEI-busmaster, 2017-10-30:
Issue is fixed with v3.2.2

closed 2017-10-30 by RBEI-busmaster