rbei-etas/busmaster #1104: DBC to DBF conversion error 3.2.1

arthur-stephenson, 2017-09-11:
BusMater 3.2.1
Windows 10

I have dbc file with
`BO_ 2158845952 ODIAction_CntrStack_LS: 8 RadioHead_LS
 SG_ ODIAC_FUCID : 15|8@0+ (1,0) [0|0] ""  SBZA_L_LS,FCM_LS,PDIM_LS,EOCM_F_LS,UPA_LS,BCM_LS,IPC_LS,ECC_LS,MSM_LS,FCP_LS,APA_LS,UHP_Low,ONSTAR_LS
 SG_ ODIAC_DspMID : 31|8@0+ (1,0) [0|0] ""  SBZA_L_LS,FCM_LS,PDIM_LS,EOCM_F_LS,UPA_LS,BCM_LS,IPC_LS,ECC_LS,MSM_LS,FCP_LS,APA_LS,UHP_Low,ONSTAR_LS
`
 I convert it to dbf
`[START_MSG] ODIAction_CntrStack_LS,11362304,8,5,0,X,RadioHead_LS
[START_SIGNALS] ODIAC_FUCID,8,2,0,U,255,0,0,0,1,,,SBZA_L_LS,FCM_LS,PDIM_LS,EOCM_F_LS,UPA_LS,BCM_LS,IPC_LS,ECC_LS,MSM_LS,FCP_LS,APA_LS,UHP_Low,ONSTAR_LS
`

from CanDB++ I see the message ID should be x10AD6000
from the datalogging from VehicleSpy i see
ODIAction_CntrStack_LS | SW CAN | RadioHead_LS | 10AD6080

Busmaster has the wrong message ID (is the priority set wrong??)
If I manually update the dbf message with the correct ID  
`[START_MSG] ODIAction_CntrStack_LS,279797888,8,5,1,X`

Then BusMaster decode the message correctly

RBEI-busmaster, 2017-09-13:
BusMaster database for J1939 saves the PGN during conversion

arthur-stephenson, 2017-09-14:
This is from automotive (GMLAN) single wire can database. I do not think it is J1939 complaint.
I am not using J1939 hardware to log (PEAK PCAN) so the hardware layer is not J939. Am I converting it as J1939?