rbei-etas/busmaster #1106: Cannot build simulation Node: mingw32-make.exe: *** [test.o] Error -1073741502

arthur-stephenson, 2017-09-20:
Hello
I am running
Win10
BUSMASTER 3.2.1
tdm-gcc-4.8.1 install here: C:\TDM-GCC-32

I get 
mingw32-make.exe: *** [test.o] Error -1073741502

Is there an additional library I need to add?

RBEI-busmaster, 2017-09-25:
Could you please execute 'gcc -v' in command prompt and share the screen shot.

arthur-stephenson, 2017-09-25:
output from

`c:\tmp\2delete6>gcc -v > gccv.txt
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=c:/tdm-gcc-32/bin/../libexec/gcc/mingw32/4.8.1/lto-wrapper.exe
Target: mingw32
Configured with: ../../../src/gcc-4.8.1/configure --build=mingw32 --enable-languages=ada,c,c++,fortran,lto,objc,obj-c++ --enable-libgomp --enable-lto --enable-graphite --enable-libstdcxx-debug --enable-threads=posix --enable-version-specific-runtime-libs --enable-fully-dynamic-string --enable-libstdcxx-threads --enable-libstdcxx-time --with-gnu-ld --disable-werror --disable-nls --disable-win32-registry --disable-symvers --enable-cxx-flags='-fno-function-sections -fno-data-sections -DWINPTHREAD_STATIC' --prefix=/mingw32tdm --with-local-prefix=/mingw32tdm --with-pkgversion=tdm-1 --enable-sjlj-exceptions --with-bugurl=http://tdm-gcc.tdragon.net/bugs
Thread model: posix
gcc version 4.8.1 (tdm-1)`

arthur-stephenson, 2017-09-25:
Thank you for the help!
Regards
Arthur Stephenson

ellis-me2, 2019-03-07:
Hi Arthur,

I'm getting the same issue you described.  Did you get a solution?

Thanks,
Matt Ellis

arthur-stephenson, 2019-03-09:
Hello Matt
I did not. I have not used BusMaster in a few months.

Regard
Arthur Stephenson
Bosch Car Multimedia

ellis-me2, 2019-03-12:
Hi Arthur,

Figured this out - at least for my system!  I had build tools from the yagarto project installed because they were required for a different project I was working on.  I uninstalled those tools and then was able to build the Busmaster project.  Note that I had the TDM GCC path before the yagarto tool in my path variable but that didn't matter - I had to actually uninstall the yagarto tools to get this to work.

Perhaps you'll be able to build as well if you uninstall all other build tools that might conflict with TDM.

Regards,
Matt