rbei-etas/busmaster #1112: busmaster don't support kvaser  lin 2.0

gpulsex, 2017-10-13:
I choose  kvaser lin  connect to  busmaster。but  all frames checksum  is classic。
I  think busmaster don‘t supports LIN enhanced checksun。 I hope you can Adopt the question，and Give a new version as soon as possible。
Important things are to be repeated for 3 times：
busmasterdon‘t supports   kvaser  LIN enhanced checksun！
busmasterdon‘t supports   kvaser  LIN enhanced checksun！
busmasterdon‘t supports   kvaser  LIN enhanced checksun！

ARL2OFO, 2017-11-08:
Hi at all, Hi gpulsex,
I can confirm I have the same issue; With the cable Kvaser LIN is not possible to send message as a master device over the LIN bus,  with the enhanced checksum: I tried both using an LDF > 2.0 or overwriting the LDF setting by a manual setting.

Stefanuccio, 2018-05-24:
Hello , 

I've encountered the same problem . Are there any update about this topic?

Thank you

ARL2OFO, 2018-05-28:
Hello,

No, I have not any news about.

I hope, the issue will be solved by the software group.

Cordiali Saluti / Best regards,

Luca Arena

Tel. +39 0373 249 735

TianLiJun, 2018-11-23:
Hello，
I have the same issue. Anyone sloved it ? 
Thank you

JanSiroky778, 2021-11-05:
Hello.
Same for me. 

Forced 2.1 LIN version.
[image]

BUSMASTER LDF editor correctly indicates that enhanced checksum will be used.
[image]

When BUSMASTER runs, there is no way to generate enhanced checskum. Message window.
[image]

It is not only error of checksum type displaying. When i use classic checksum on my LIN slave HW unit, that is communicating with BUSMASTER, the communication goes well. 

my setup:
Busmaster v 3.2.2
KVASER Leaf Profesional LIN (firmware updated, driver updated)