rbei-etas/busmaster #1113: Issue with Test Automation execution

JayasagarC, 2017-10-24:
Created a Send Request CAN Frame in Test Automation executor when the case is executed using the Test suite executor, the data in the frame is changing dyanamically
for: in Dbf created a CAN Frame 0x740 and payload as Data00 8bits startbit 0 Byteindex 0 little endian, Data01 8bits startbit 0 Byteindex 1 little endian.... till Data07 8bits startbit0 Byteindex 7 little endian.
created a test case from Dbf file and data as 0x04 0x22 0xf1 0x86.
when the above test case is executed using Test suite executor the data in payload is changing dynamically across the different executions

JayasagarC, 2017-10-25:
@RBEI-busmaster

itiserik, 2017-10-30:
+1 for me here. When I try to send something using the test automation executor it seems that it sums up the values of the payload and that is send as byte[0] in the can message.

Can you show us the code for this version (GPL...)?

itiserik, 2017-10-30:
Tried to debug somewhat into it:
In TestSuiteExecutorGUI/TSExecutionCAN.cpp:596 the function GetSignalList is called. On every invocation of the test this function returns (into signalList) another list of signals with a changed order. (The ascending pointer determines the order)

Observing the code it is assumed that signalList has the same order as ouSendData.m_odSignalDataList

The function GetSignalList is probably defined in DBManager.dll on a private github somewhere.

I looked into TXWindow/CANTxMsgItem.cpp: 363 and there a sortByStartBit function is used to keep the order consistent. A trick like this should be performed on TSExecutionCAN.cpp...

itiserik, 2017-11-01:
See #1116

JayasagarC, 2017-11-02:
I have not written any as such for above testing
followed below steps:
1. created data base file with 2 msgs with signals as below
 MsgTx-> 0x7df 8 bytes and signals as Data00(8bits), Data01(8bits), Data02(8bits), Data03(8bits), Data04(8bits), Data05(8bits), Data05(8bits), Data07(8bits)
 MsgRx-> 0x748 8 bytes and signals as Data00(8bits), Data01(8bits), Data02(8bits), Data03(8bits), Data04(8bits), Data05(8bits), Data05(8bits), Data07(8bits)
2. Using Test Automation Editor created Test case as below
   Test Case1
    Send 
      MsgTx
       Data00 -- 0x04
       Data01 -- 0x22
       Data02 -- 0xf1
       Data03 -- 0x86
       Data04 -- 0x00
       Data05 -- 0x00 
       Data06 -- 0x00
       Data07 -- 0x00 
    Verify
     MsgRx
      Data00 -- 0x00
      Data01 -- 0x00
      Data02 -- 0x00
      Data03 -- 0x00
      Data04 -- 0x00
      Data05 -- 0x00
      Data06 -- 0x00
      Data07 -- 0x00
3. Using Test Automation Editor executed above Test Case.
 The Result is each time i execute the case send data is not as expected
 data is getting modified randomly.

itiserik, 2017-11-02:
@JayasagarC take a look at my patch, on my desk it solves these randomly data issues.

JayasagarC, 2017-11-09:
could you please help me in taking the update in busmaster installation exe

JayasagarC, 2017-11-15:
Is the issue fixed in 3.2.2 version?

closed 2017-12-11 by JayasagarC

xzp11, 2023-02-07:
@itiserik hello,could you please help me in taking the update in busmaster installation exe or maybe it is a dll files.I just find source code in your place，but I don't know how to update the busmaster software. Thank you very much.

xzp11, 2023-02-10:
if  anybody need,you can get a TestSuiteExecutorGUI.dll files in my repository.it can be solve above problem.