rbei-etas/busmaster #1119: busmaster3.2.2 with blf conversion to log

junghoobae, 2017-11-20:
hello I'm junghoo bae from ETAS korea.

what I found is blf to log conversion tool is not work collectly .
ID and Data in log file is fine but went it replayed everything are different from blf file.

if you need more info I will send you.
you can find my name in bosch connect.

robert-16, 2019-03-22:
Hello all,

I have experienced similar issue on BUSMASTER 3.2.2:
Format conversion tool states "Error: No CAN data found for conversion." when trying to convert .BLF file from CANoe 10.0 to a .LOG. It means the output result is an empty (0KB) .LOG file.

[image]

Thank you in advance for your support.

Best regards

Robert

cricape06, 2021-06-24:
Same Issue here, Did you manage to find a workaround?

Thanks!

robert-16, 2021-06-25:
Hello cricape06, Hello all,

no, I did not find any workaround other than using Vector CANoe to analyse the .BLF log file.

Best regards

Robert.

cricape06, 2021-09-04:
Hi Robert, thanks for your reply. I managed to convert it using the converter tool from Kvaser (no issues).

Best regards,
Cristian