rbei-etas/busmaster #1120: SetTimeout function not working

JimmyC1984, 2017-11-27:
Hi,

I try to use the SetTimeout API function in node simulation, however when I try to build the file the compiler tells me SetTimeout is not declared. 
Does anyone have the same issue?
Thank you!

nostar, 2017-11-27:
There is no SetTimeout() function in the CAN node sim API.  Do you mean SetTimerVal()?
```
GCC_EXTERN /*extern "C"*/ BOOL GCC_EXPORT SetTimerVal(char*,UINT);
```

JimmyC1984, 2017-11-27:
Hi,

Sorry didn't mention it clearly, the SetTimeout API is in J1939 CAN application.
When I use the Node simulation, I insert the SetTimeout function and during the build, the compiler shows SetTimeout not declared.
Thank you!

nostar, 2017-11-28:
Ah, I see.  Looks like they typo'd the function name in the header file https://github.com/rbei-etas/busmaster/blob/master/Sources/BUSMASTER/Application/SimulatedSystems/include/BMJ1939Defines.h

```
GCC_EXTERN UINT GCC_EXPORT SetTimout(BYTE, UINT);
```

If your comfortable with the Busmaster source, you can fix the typo in the header and rebuild LibWrapper_J1939.a and replace the installed version, using the batch script in /Sources/BUSMASTER/Application/SimulatedSystems/include/ .  Otherwise, you are outta luck untill they fix it.