rbei-etas/busmaster #1123: Does Busmaster support two CAN node simulation?

JimmyC1984, 2017-12-04:
Hi,

I'm using Busmaster with 2 peak CAN adapter, 1st PeakCAN adapter is on J1939 CAN and 2nd PeakCAN adapter is on standard CAN.
I'm trying to set up an automatic testing platform, J1939 is used to send some command and read command back from standard CAN, could the code in J1939 node simulation read the RxMsg feedback from the standard CAN node simulation?
Thank you!

nostar, 2017-12-05:
You can create and run as many node simulations as you want, but no method exists in Busmaster to pass data in between simulated nodes.  In Vector products (CANoe/CANalyzer) this is done via environment variables or system variables.  

See my issue https://github.com/rbei-etas/busmaster/issues/839 . I needed to share data between a CAN node and a LIN node.  Since MinGW is used to compile node simulations, you have the entire Windows API at your disposal.  I ended up creating a shared memory segment as shown in the last comment, which makes ' lpvMem' a pointer to memory available to all nodes that use that code.  You also have to put the following lines in the global variable section of each node simulation:
```
static LPVOID lpvMem = NULL;
static HANDLE hMapObject = NULL;
BOOL fInit, fIgnore;
```

JimmyC1984, 2017-12-05:
It works perfectly! Thanks!