rbei-etas/busmaster #1126: Support for Sontheim Vehicle Communication Interfaces

jogo-, 2017-12-09:
Sontheim have a wide range of CAN "Mobile Interfaces & VCIs" already supported by other BOSCH product:
GRADE-X Authoring Environment Copyright (c) Bosch Automotive Service Solution Ltd./GmbH
(Stockport ENGLAND / Garching GERMANY)

So, at least 2 of these hardwares (CANUSB & COMfalcon) are already available at BOSCH BASS.
Is there any plan for supporting Sontheim VCIs in RBEI-ETAS BUSMASTER ?

References:
https://www.sontheim-industrie-elektronik.de/en/products/automotive/interfaces-gateways/mobile-interfaces-vcis/

itiserik, 2017-12-11:
I started to implement support for Sontheims CANfox in BUSMASTER. I can push this to my forked repo.

jogo-, 2017-12-21:
That would be great and should help us a lot.
Many thanks for your contribution.

jogo-, 2018-05-26:
@itiserik Have you made any progress in implementing support for Sontheims CANfox in BUSMASTER ? Can you please push it to your forked repo or even do a PR here ? Thx

jogo-, 2019-02-09:
@itiserik Have you made any progress in implementing support for Sontheims CANfox in BUSMASTER ? Can you please push it to your forked repo or even do a PR here ? Thx once again

quantum8, 2019-07-16:
I'd like for busmaster to support these adapters too.

I've got a CANfox adapter from IFM that identifies in device manager as being from Sontheim Industrie Elektronik GmbH.