rbei-etas/busmaster #1127: No info in Message window - need to restart

Teddyz, 2017-12-09:
After some use, I suppose including a glitch in communication with my CancaseXL Log or just me swapping to another CAN interface, I see no messages in the Message window.  A simulated node works, output to Trace works, keyboard input works.

Restarting (Pressing connect) does not help. Only solution I have found is to restart the program.
Seen in 3.2.1, 3.2.2. I have not tested earlier versions.

RBEI-busmaster, 2017-12-12:
Could you please share the log file. Please specify the steps to reproduce the issue.

GuillaumeFege, 2018-01-24:
To reproduce it :  
- Select a CAN HARDWARE
- Connect to it
--> Frame are correctly shown in "Message Window"
- Disconnect
- Select an other CAN HARDWARE
- Connect to it
--> Nothing is shown in "Message Window"
But, frame are correctly wrote to the log file

Version 3.2.2

Teddyz, 2018-01-25:
After testing some more:
-This can happen even if no simulation is active.
-When this happen, the Main Diagnostic Window does not receive any Responce Data. (Something with filters???)
-The problem always start from when I press Connect (i.e. not sporadic). Or, of course, previous Disconnect, but that is impossible to know.

Teddyz, 2018-04-26:
Also, when this happens, it is impossible to send diagnostics that are longer than 7 bytes.