rbei-etas/busmaster #1129: void OnBus_Connect() is not run first

Teddyz, 2017-12-09:
I have a Bus event routine that is run on Connect. It makes two blank lines in the Trace window and some more. 
From version 3.2.2 this code is not run first. Instead some other messages are handled before this routine.

Minor issue.

RBEI-busmaster, 2017-12-12:
Please share the configuration along with the .cpp and .dbf files for further analysis

Teddyz, 2017-12-12:
Added configuration and a simple test-node. No database needed. To test, have another node transmitting a fram periodically at 100ms.
[Busmaster.zip](https://github.com/rbei-etas/busmaster/files/1550871/Busmaster.zip)

I noted that when simulation is not running, the CANcaseXL does not ACK frames. I do not know if this is intended or not.

GuillaumeHM, 2018-04-04:
I faced the same problem with same timer of 100ms. Its callback is called before Connect event on my version 3.2.0.
You may use OnDLL_Load instead of Connect event.