rbei-etas/busmaster #1133: Signal Description on view

Secmick, 2017-12-26:
Hi,

I try to show on "Message Window" Output Signal description from DBF (DBC)
[image]

Version: 3.2.2

ameliip, 2018-01-30:
Good morning,
I wonder if this will ever be solved. 

Moreover, i don't see it as a "minor improvement", as it was working in the old version.
"Bugfix" would be more appropriate. For a lot of people working in the automotive field this is a key feature, so i kindly ask @RBEI-busmaster to kindly provide a feedback on this.

Thanks,
Pascal.

MNY-Dinex, 2020-01-30:
@RBEI-busmaster . Any news regarding this function? I really would like this to work. It is essentiel for our use of Busmaster.

Regards
Mads Nygaard

Secmick, 2020-11-18:
@RBEI-busmaster Any update for this issue/Improvement? is this issue going to be fixed?