rbei-etas/busmaster #1140: #problem with busmaster compiling with MinGW

Andreal80, 2018-01-16:
I had a problem using node simulation.
I actually installed tdm-gcc-4.8.1 following the help guide.

this is the output from the bulid log:

[image]

this is my current version of MinGW: 
[image]

Many thanks.

GuillaumeHM, 2018-04-04:
Busmaster is very sensible to desynchronization on multiple subjects :
1. PATH variable to select the right mingw compiler if multiple installed.
2. 32/64 bits version of windows/busmaster/mingw
3. Database cache files at C:\ProgramData\BUSMASTER\General\DBCache. They are generated when you associate a dbf to your project.
4. node.cpp and node.def

I think your problem is around the fourth point. Try to use the Busmaster code editor to (re)declare (meaning resynchronize) your functions/events using the left side panel and don't forget to save def file with cpp one. Other files are generated from those two.

With hope to help you.

excision10, 2018-11-05:
But, how do I create the .def file?

hugoquiroz, 2019-05-07:
Thanks! I was having issue with no. 3, however I had to dissociate, and the re-generate the dbf as well

hugoquiroz, 2019-05-07:
> But, how do I create the .def file? It is not generated automatically.

I was able to generate when you open the .cpp in busmaster, save it, then compile again.