rbei-etas/busmaster #1147: I tried to convert a .CAN file to .CPP, then it was generated with a set of errors

islamelhariry, 2018-01-29:
When I tried to convert the "Client.can" file to .cpp file, it gave me the errors in the attached file "Clientlog", after that when I tried to build the generated "Client.cpp" file it gave me the errors in the attached file "error" & "output".
Kindly check these errors, I hope you can help if you have any clue why these errors took place.
[Clientlog.txt](https://github.com/rbei-etas/busmaster/files/1672487/Clientlog.txt)
[error.txt](https://github.com/rbei-etas/busmaster/files/1672488/error.txt)
[output.txt](https://github.com/rbei-etas/busmaster/files/1672489/output.txt)

Ravenstone0910, 2018-02-22:
I think there are two different problems. 
Busmaster is not able to convert all availaible functions from CAPL-files. I don't know CAPL very good, but as I can see in you ClientLog.txt, you can use Winsockets for TCP and UDP connections, which you have to implement manually.
"'mingw32-g++' is not recognized as an internal or external command," looks like that you didnt't set up the compiler correctly.

islamelhariry, 2018-02-25:
I tried another file without TCP and UDP connections and the tool displayed similar errors, kindly check it.
For the installation of the MinGW compiler I installed it  with the same version and at the same destination as described in the help.
[image]

[error.txt](https://github.com/rbei-etas/busmaster/files/1755925/error.txt)
[output.txt](https://github.com/rbei-etas/busmaster/files/1755926/output.txt)
[testVS6log.txt](https://github.com/rbei-etas/busmaster/files/1755927/testVS6log.txt)