rbei-etas/busmaster #1148: Works with CANopen?

jay1956, 2018-02-12:
Hi,
I´m using Busmaster for communication with a Battery Module. The Battery Module uses CanOpen (CIA418 & CIA 454) and has a static ID. The first 4 Bytes of the 8 Byte Data field are used for defining the transmitted data and the other 4 Bytes contain the data.

Now the Question is how I can filter on the Message content. I found only the possibility to set a filter on the ID but I need to filter on the first 4 Data Bytes. 
The same question is for defining the Messages in the Database editor. I need to define different Signals depending on the first Data Bytes.

FrankGerlachFlex, 2019-07-31:
You could probably code your own Message Filter inside a Node. It should not bee too hard to decode the CANopen message for the information you need. If you know C++, of course.

votuananhs, 2022-12-22:
@jay1956 You can connect CANopen device to Busmaster software tool. Is this right ? If yes, please share with me how to do that. Because i didn't see that hardware device from latest version for Busmaster. Many thanks in advance !