rbei-etas/busmaster #1154: 3.2.2: error while parsing (error : unexpected token found at line 7)

fabioint, 2018-03-22:
Hello,
I try to create a new ldf file, but I'm not able to do it.
When I try to open the ldf I received the error in the object.

Best regards,
Fabio

RBEI-busmaster, 2018-03-22:
Please share the below details:
1. BUSMASTER version used
2. Send the LDF file which you are trying to open to BUSMASTER@in.bosch.com

fabioint, 2018-03-22:
I'm using the 3.2.2 version.
I create new 2.0 LDF file. If I try to reopen the file, I receive the error.

Thank you very much for your support.
Fabio

RBEI-busmaster, 2018-03-26:
Thanks for the steps.
We are able to reproduce the issue.

fabioint, 2018-03-26:
Sorry, but I didn't understand if you provide a patch.

Thanks,
Fabio

thehien377, 2020-09-01:
Hello,

I got the same issue?
Do you have any fixing?

Best regards,
HienBui

fffireflyff, 2020-12-01:
Look like these issues and more were created by BOSCH purposely!