rbei-etas/busmaster #1157: BUSMASTER switch to RERO Development

jogo-, 2018-03-28:
May I suggest you to adopt this pretty simple release philosophy for BUSMASTER : **Release Early, Release Often (or RERO)**
https://en.wikipedia.org/wiki/Release_early,_release_often

The last BUSMASTER v3.2.2 was released 5 months ago on Oct 30, 2017.
Would you mind please sharing with us your schedule for the next release ?

Many thanks.

ZeroAviation, 2018-04-20:
I love Busmaster, it is by far the best CAN tool I have used so far. However, I am sad that the development seems to have dropped off :(

Would love to hear what ETAS or RBEI has to say about its future!

gxurma, 2018-04-21:
And I am also curious, if there will ever be a multiplexed signal interpretation...

jogo-, 2018-05-23:
@RBEI-busmaster What is exactly the current status of this RBEI BUSMASTER project ? It seems to be a bit frozen atm... I hope it's still alive :-)

ETAS-Nithin, 2018-05-25:
@jogo- The Busmaster development team’s “wish-list” for usability & functionality features keep getting bigger based on constant feedback from the users. :)

Since we haven’t had enough helping hands via code contributions, implementing a lot of them with a small team was quite an effort. Hence the focus was to prioritize stability, fix any major bugs and try to implement the less effort-intensive features over a period of time.

ETAS-Nithin, 2018-05-25:
With the release of v3.2.x, idea was to ensure stability on Win10 and the plug-in architecture that allows users to create plug-in extensions to use with open source tool. This is the reason for no new releases since v3.2.2, but the project is very much alive :)

And with contributions from keen community developers, it would be possible to implement the wish-list much faster.

yegorich, 2018-05-25:
@ETAS-Nithin Thanks for an update. I'm glad to hear that the project is still alive. But one cannot see the progress from outside as you don't push single commits. These big commits with thousands of changes are very contra productive for the people willing to contribute. Besides I see 10 almost untouched PRs.

Making CMake working again and enabling other Visual Studio versions would be also very helpful.

AppVeyor support would be very much appreciated.

jogo-, 2018-05-26:
@ETAS-Nithin I totally agree with @yegorich suggestions

> Making CMake working again and enabling other Visual Studio versions would be also very helpful.
> AppVeyor support would be very much appreciated.

This will certainly and definitely help in getting more contributions from the community developers.

jogo-, 2018-06-30:
@ETAS-Nithin & @RBEI-busmaster The last BUSMASTER v3.2.2 was now released 8 months ago on Oct 30, 2017 #652

> Since we haven’t had enough helping hands via code contributions, implementing a lot of them with a small team was quite an effort. Hence the focus was to prioritize stability, fix any major bugs and try to implement the less effort-intensive features over a period of time.

Even with a RBEI "small team" from a giant company as BOSCH, it should be possible achieve bigger things than just "**try** to implement the **less effort-intensive features**".
I'm used to work with BOSCH BASS and even if it's not always easy every day, we are doing a lot more in a same "period of time".

> And with contributions from keen community developers, it would be possible to implement the wish-list much faster.

You could first try to make the most of the current 10  PRs to prove us you care about "contributions from keen community developers" and maybe it could be the signal we are all waiting for to start a new round of contributions...

jogo-, 2019-02-06:
@ETAS-Nithin & @RBEI-busmaster AGAIN: What is exactly the current status of this RBEI BUSMASTER project ? It seems to be Totally frozen atm... I hope it's still alive :-)