rbei-etas/busmaster #1163: problem busmaster

MERYEM321, 2018-04-12:
hello,
how can I convert dbc to dbf on busmaster version 3.2.1, because i found a problem

RBEI-busmaster, 2018-04-13:
Format Converters (available under 'Tools' tab) can be used for DBC to DBF conversion. 
DBC to DBF Conversion option is available in the 'Other Converters...' tab.

MERYEM321, 2018-04-21:
hi
I want to send a frame and after 100ms I want to change the value of a
frame signal
how can I do that using node simulation
thanks

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

ETAS-Nithin, 2018-05-07:
@MERYEM321 if the file conversion issue is resolved, we would close this issue. Request to post the node sim query as a new topic for better solution from the community. thanks

closed 2018-05-07 by ETAS-Nithin